douglasjunior/AndroidBluetoothLibrary #1: is there auto connection ?

pranavlathigara, 2017-05-31:
Hi
Thank you for this Library 
I have question about this Library 
Is there any option for auto connection for Specific Device after Loss Connection ?

douglasjunior, 2017-05-31:
The library does not, but you can do it easily. Something like:

```java
    @Override
    public void onStatusChange(BluetoothStatus status) {
         if (lastStatus == BluetoothStatus.CONNECTED && status == BluetoothStatus.NONE) {
              mService.connect(device);
         }
         lastStatus = status;
    }
```

You can also open a dialog indicating that you are reconnecting, etc.

douglasjunior, 2017-05-31:
You can also connect to a device knowing your mac.

```java
BluetoothDevice device = mBluetoothAdapter.getRemoteDevice(macAddress);
mService.connect(device);
```

pranavlathigara, 2017-06-01:
yes this is nice but once i lost connection or out of range with specific device and then come to range it gets auto connect with that device 
is it possible..?

douglasjunior, 2017-06-01:
Yes, you can create a service that keeps trying to connect again.

douglasjunior, 2017-06-12:
Please, if you have no further questions, you can close the issue. 😄

closed 2017-06-25 by douglasjunior