douglasjunior/AndroidBluetoothLibrary #2: Message Queueing

kioko, 2017-06-06:
How can I wait for a response before running another command? I.e I have a list of 3 commands, before running the second command, I wait for a response from the first one.

douglasjunior, 2017-06-06:
You need to control this because communication is asynchronous. Bluetooth does not depend on you sending information to return an response.

Every time the device sends a response it will be caught by the event `onDataRead`.

```java
service.setOnEventCallback(new BluetoothService.OnBluetoothEventCallback() {
    @Override
    public void onDataRead(byte[] buffer, int length) {
         // If you want to convert to String
        String text = new String(buffer, 0, length);
        sendNextCommand();
    }
    ...
});

private void sendNextCommand() {
      bytes[] nextCommand = ...;
      service.write(nextCommand );
}
```

douglasjunior, 2017-06-07:
Please, if you have no further questions, you can close the issue. 😄

closed 2017-06-07 by kioko