douglasjunior/AndroidBluetoothLibrary #3: Nao consigo recepcionar nada no public void onDataRead(byte[] buffer, int length) {

a9336, 2017-07-03:
Boas, tudo bem?
Olha o meu objectivo era fazer um pedido a um servidor remoto, que responde de imediato, enviando uma trama, e receber a resposta.
Testei com um terminal e funciona, mas desta forma nao.

Enviar, sem problemas, receber é que nao

 service.setOnEventCallback(new BluetoothService.OnBluetoothEventCallback() {
            @Override
            public void onDataRead(byte[] buffer, int length) {
                String text = new String(buffer, 0, length);
                Toast.makeText(getApplication().getBaseContext(), text, Toast.LENGTH_SHORT).show();
            }

 public void enviar(View view) throws UnsupportedEncodingException {

        writer = new BluetoothWriter(service);
...
writer.writeln(joined);
}

podes ajudar?

douglasjunior, 2017-07-03:
Olá @a9336 

O evento `onDataRead` é disparado em duas situações:
1. Quando se recebe o caractere delimitador. Ex: `characterDelimiter = '\n'`.

   Neste caso, a biblioteca armazena em buffer todos os caracteres recebidos, e só chama o `onDataRead` quando um caractere `\n` for recebido.

2. Quando se completa o tamanho do buffer especificado. Ex: `bufferSize = 1024`.

   Já nesta situação, se nenhum caractere delimitador for recebido, a biblioteca armazena os dados recebido até some 1024 bytes, então o evento `onDataRead` será disparado.

Esta é uma forma de otimizar o recebimento de dados e se ganhar em performance.

a9336, 2017-07-04:
muito obrigado, estou a receber de um raspberry a informaçao.

a trama esta a ser enviada depois de passar por um conversor (binascii.b2a_uu(data)
Convert binary data to a line of ASCII characters, the return value is the converted line, including a newline char. The length of data should be at most 45.)

agora falta so converter a informaçao no android, de resto ok ;)

obrigado

douglasjunior, 2017-07-04:
De nada! 👍

closed 2017-07-09 by douglasjunior