douglasjunior/AndroidBluetoothLibrary #4: Status Connect

Windows87, 2017-07-24:
Olá,
 Estou tentando conectar a um dispositivo BLE (consigo escanear e adicionar na RecyclerView), quando clico nele na RecyclerView só uma vez, ele muda de status para Connecting e depois None, e só depois que eu clico três ou mais vezes seguidas, ele conecta (o led do Bluetooth fica normal em vez de piscando), mas mesmo ele conectando, não retorna o status como Connected.

douglasjunior, 2017-07-24:
Você já identificou o UUID do `service` e da `charactiristic`? Qual é o modelo do smartphone e do dispositivo BLE?

Copie aqui o código que você usou para iniciar a biblioteca.

douglasjunior, 2017-07-24:
Eu já tive a experiência de alguns smartphones não serem capaz de conectar à alguns dispositivos BLE, não sei se é Bug do Android ou alguma incompatibilidade no hardware.

O que costumo fazer é utilizar programas para testar a conexão e descobrir o UUID correto do `service` e da `characteristic`:
- https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan&hl=pt_BR
- https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan&hl=pt_BR

Windows87, 2017-07-24:
Eu troquei o UUID do service e do characteristic do meu dispositivo BLE (HM-10), e consegui conectar, mas não tem alguma maneira de que ao quando clicar para conectar o dispositivo, o aplicativo já descobre a UUID do Service e do Characteristic e aplicar automaticamente eles na config?

douglasjunior, 2017-07-24:
Não é garantido, pois em alguns dispositivos ele pode aparecer mais de um `service` e `characteristic` que aceitam leitura/escrita via conexão serial, mas nem todos funcionam. O melhor é você saber quais dispositivos você vai conectar e já deixar programado os UUIDs de cada um.

Eu já trabalhei com o HM-10, tive muitos problemas.

douglasjunior, 2017-07-24:
**Corrigindo...**

Não é garantido, pois em alguns dispositivos ele pode aparecer mais de um `service` e `characteristic` que aceitam leitura/escrita via conexão serial, mas nem todos funcionam. O melhor é você saber quais dispositivos você vai conectar e já deixar programado os UUIDs de cada um.

Eu já trabalhei com o HM-10, não tive tive muitos problemas.

Windows87, 2017-07-24:
Está funcionando perfeitamente agora, dá pra fazer read, write e conectar.
Dúvida resolvida, obrigado.

closed 2017-07-24 by Windows87