douglasjunior/AndroidBluetoothLibrary #7: D is set to true in BluetoothService.

medvednic, 2017-08-20:
The `D` property is set to `true` in `BluetoothService`, it results in debug log messages always being produced when using `BluetoothClassicService`.
The library currently can't be used for production since google forbids logging in release grade applications.

douglasjunior, 2017-08-20:
When you compile your app in production mode, the logs are ignored by the operating system, that makes no difference.

medvednic, 2017-08-20:
Maybe they are, but google explicitly says not to include `Log.*` lines in release applications.

douglasjunior, 2017-08-21:
Curious, I have several projects in production and I have never worried about it.

Would you have the reference for this recommendation? I would like to read more about it.

medvednic, 2017-08-21:
In the official [user guide](https://developer.android.com/studio/publish/preparing.html#publishing-configure).