douglasjunior/AndroidBluetoothLibrary #8: There is a problem connecting to iOS/OSX BLE.

ShockUtility, 2017-09-17:
There is a problem connecting to iOS/OSX BLE.
Connection is completed and writing is ok.
However, onDataRead does not work.
I did various tests and found a solution.
But, I do not know why because I do not know about Bluetooth.

--- BluetoothLeService.java 200 line ---
`
boolean isOK = gatt.setCharacteristicNotification(characteristic, true);
if (isOK) {
     for (BluetoothGattDescriptor descriptor : characteristic.getDescriptors()) { 
          descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
          gatt.writeDescriptor(descriptor);
     }
}
`

douglasjunior, 2017-09-17:
What information do you expect to receive from `onDataRead`? Do you have a character to indicate the end of the data?

If is not, the library will store the received data until it completes the `Buffer` size, after completing the `Buffer`, the content is dispatched by `onDataRead`.

You can configure your delimiter character as well as the `Buffer` size.

```java
config.bufferSize = 1024; // Number of characters stored in the buffer until it is dispatched.
config.characterDelimiter = 'your-char-here'; //character that indicates the end of the data, whenever found the buffer is dispatched
```

ShockUtility, 2017-09-18:
You're right.
It was my mistake.
But the code I showed is worth considering for accessing Apple's Bluetooth.

This code affects the behavior of "[didSubscribeTo](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanagerdelegate/1393261-peripheralmanager)" and "[didUnsubscribeFrom](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanagerdelegate/1393289-peripheralmanager)" in "[CBPeripheralManagerDelegate](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanagerdelegate)" in "iOS / OSX".

douglasjunior, 2017-09-26:
Sorry, I'm not familiar with communication between Android and iOS via BLE. The idea behind this library is to work with a communication pattern, to avoid the complex manipulation of strings when receiving data. If your data dont follow a pattern, then perhaps you would have to manipulate the BLE natively.

closed 2017-12-12 by douglasjunior