douglasjunior/AndroidBluetoothLibrary #17: Unable to send message.

sswxyz17, 2018-03-27:
I am trying to send a message over BLE, the devices are connected to each other but the message is not sent. I think the issue is this: 
E/BluetoothLeService: Could not find uuidService:e7810a71-73ae-499d-8c15-faa9aef0c3f2 and uuidCharacteristic:bef8d6c9-9c21-4c9e-b632-bd58c1009f9f
but I don't understand what to do with it.

P.S 
I am just getting a list of devices and then I user the normal connection method(service.connect()) to connect with one of them.

douglasjunior, 2018-03-28:
First you need to make sure that you have selected the correct UUID for the `service` and for the `characteristic`. Then, you can `write` after the connection has been established successfully.

You can use this apps to help to find the UUIDs:

- https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan
- https://play.google.com/store/apps/details?id=com.edodm85.bluetoothbleterminal.free

sswxyz17, 2018-03-28:
Thank you for the answer, It helped me solve the issue. I have one other small question. 
Is is possible to read(message) from a peripheral, for example when I click a button or do I need to wait for the peripheral to send the data(message)?

closed 2018-04-09 by sswxyz17