douglasjunior/AndroidBluetoothLibrary #20: Reading on demand.

sswxyz17, 2018-04-09:
Hello, sorry if this is a stupid question but I was wondering if there is a way to read from a peripheral device on demand, like for example when I press a button or something else? As far as I understand you can read when a device writes to you, but I was still wondering if there is another way to read.
Thank you in advance.

douglasjunior, 2018-04-10:
The remote device needs to write something for you to read.

You can write something to the remote device, and then the remote device read the message and writes something back.

ederdoski, 2018-06-05:
Hello, If I wanted to read on demand, would not there be any solution that does not include writing to the device?

jjrodcast, 2018-06-05:
@ederdoski you mean that you want to read when you fires an action? If that's the case I think you have to send the corresponding data to other devices.

jjrodcast, 2018-06-07:
@ederdoski it is simple, when you want to read data there is an interface  where you can read the bytes the same if you want to send data to other device.

closed 2018-12-15 by douglasjunior