douglasjunior/AndroidBluetoothLibrary #21: Bluetooth disconnect from device when app closes

gilgoldzweig, 2018-05-07:
The service disconnect without giving the option to control when to disconnect the device.

There is no way for me to override when you disconnect from the device.

douglasjunior, 2018-05-07:
This is a default bluetooth behavior, it is not the library that decides this.

gilgoldzweig, 2018-05-07:
How can I change it so that the device will not disconnect when the app close?

douglasjunior, 2018-05-07:
Something like 
https://stackoverflow.com/questions/25267753/how-to-keep-bluetooth-connection-in-background

gilgoldzweig, 2018-05-07:
I have a service that is running in the background but how can I let the service lifecycle control the BluetoothService lifecycle?

jjrodcast, 2018-05-07:
I think the service disconnect when you resume or pause your app, probably you have to make your own implementation unless I am wrong.

closed 2018-12-15 by douglasjunior