douglasjunior/AndroidBluetoothLibrary #24: not find any device

Amirhgh74, 2018-06-04:
I implement your library in Kotlin , it starts scanning but no device is found , is there anything more than implementing the scan callback ? 
 onDeviceDescover won't call , but after some times of scanning , onScanStop gets called 

I set the uuid to be null and I'm using the basic library

jjrodcast, 2018-06-05:
@Amirhgh74 I think you have to pair your phone with bluetooth devices, once it has paired the list will display in your fragment or activity. I don't know if you did it.

jjrodcast, 2018-06-05:
go to your bluetooth option in your phone, then you will see all the available bluetooths; then you have to pair with your phone.

Amirhgh74, 2018-06-06:
@jjrodcast i want to do this in my code 
I mean i want to be able to find unpaird devices and pair with theny through the code is this possible in this library ?

jjrodcast, 2018-06-06:
I don't explore that functionality yet, but @douglasjunior could answer that

thepipeman, 2018-10-10:
@Amirhgh74 were you able to fix this? I also implemented the library in Kotlin and it doesn't find any device when I do scan.

Amirhgh74, 2018-10-10:
@thepipeman no after some try I used another library 
don't know if anyone fixed this or not

douglasjunior, 2018-10-10:
Kotlin does not change anything, please, show the code that you are using to setup the library.

thepipeman, 2018-10-11:
@douglasjunior 
config below

`private fun initializeBluetoothService() {
        val config = BluetoothConfiguration()

        config.bluetoothServiceClass = BluetoothClassicService::class.java

        config.context = applicationContext
        config.bufferSize = 1024
        config.characterDelimiter = '\n'
        config.deviceName = getString(R.string.app_name)
        config.callListenersInMainThread = true

        config.uuid = UUID.randomUUID() // For Classic
        

        BluetoothService.init(config)
}`

The fragment that scans devices

`class DeviceListFragment : DialogFragment(),
        BluetoothService.OnBluetoothScanCallback,
        DeviceItemAdapter.OnAdapterItemClickListener {

    private lateinit var btAdapter: BluetoothAdapter
    private lateinit var itemAdapter: DeviceItemAdapter
    private lateinit var bluetoothService: BluetoothService
    private var isScanning = false

    override fun onCreateView(inflater: LayoutInflater, container: ViewGroup?, savedInstanceState: Bundle?): View? {
        return inflater.inflate(R.layout.fragment_device_list, container)
    }

    override fun onViewCreated(view: View?, savedInstanceState: Bundle?) {
        super.onViewCreated(view, savedInstanceState)

        btAdapter = BluetoothAdapter.getDefaultAdapter()

        itemAdapter = DeviceItemAdapter(context, btAdapter.bondedDevices)
        itemAdapter.setOnAdapterItemClickListener(this)
        device_list_rv.adapter = itemAdapter

        btn_scan.setOnClickListener{ onScanClick() }

        bluetoothService = BluetoothService.getDefaultInstance()
        bluetoothService.setOnScanCallback(this)
    }

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
    }

    override fun onItemClick(device: BluetoothDeviceDecorator, position: Int) {
        bluetoothService.connect(device.device)
    }

    // implemented methods from BluetoothService.OnBluetoothScanCallback
    override fun onStopScan() {
        isScanning = false
        pg_bar.visibility = View.GONE
        btn_scan.text = getString(R.string.action_scan)
    }

    override fun onStartScan() {
        isScanning = true
        pg_bar.visibility = View.VISIBLE
        btn_scan.text = getString(R.string.action_stop)
    }

    override fun onDeviceDiscovered(device: BluetoothDevice?, rssi: Int) {
        val dv = BluetoothDeviceDecorator(device!!, rssi)
        val index = itemAdapter.devices.indexOf(dv)

        if (index < 0) {
            itemAdapter.devices.add(dv)
            itemAdapter.notifyItemInserted(itemAdapter.devices.size - 1)
        } else {
            itemAdapter.devices[index].device = device
            itemAdapter.devices[index].rssi = rssi
            itemAdapter.notifyItemChanged(index)
        }
    }

    private fun onScanClick() {
        if (!isScanning) {
            bluetoothService.startScan()
        } else {
            bluetoothService.stopScan()
        }
    }
}`

both `onStopScan` and `onStartScan` works properly, but `onDeviceDiscovered` is never invoked.

thepipeman, 2018-10-11:
@Amirhgh74 do you mind sharing me this library? And also, did you use Kotlin as well?

douglasjunior, 2018-10-11:
@thepipeman your problem is on this line:

```kotlin
config.uuid = UUID.randomUUID() // For Classic
```

You need to provide the UUID of the target device you want to connect to, not a random one.

douglasjunior, 2018-10-11:
As I said before, Kotlin does not change the library work.

thepipeman, 2018-10-11:
@douglasjunior this is weird, because I actually used a random UUID in your sample app, and it works fine.

douglasjunior, 2018-10-11:
In the Classic mode, the UUID is used here:

https://github.com/douglasjunior/AndroidBluetoothLibrary/blob/db1df73c5c833a6c995103618dd68959e910fda7/BluetoothClassicLibrary/src/main/java/com/github/douglasjunior/bluetoothclassiclibrary/BluetoothClassicService.java#L233

See the docs: https://developer.android.com/reference/android/bluetooth/BluetoothDevice#createRfcommSocketToServiceRecord(java.util.UUID)

douglasjunior, 2018-10-11:
I created a sample project using Kotlin and converted the main classes from the previous example.

Notice that no changes were made to the library.

https://github.com/douglasjunior/AndroidBluetoothLibrary/tree/master/SampleKotlin

thepipeman, 2018-10-11:
@douglasjunior thanks man!

thepipeman, 2018-10-15:
@douglasjunior I was able to make this work, however, the problem is it only works when you have `targetSdkVersion 22` if you change this, e.g. to 26 then the library basically fails. Could you confirm this one? Thanks

closed 2018-12-15 by douglasjunior

masonlee, 2021-04-28:
In case this might help someone, on Android 11 at least, the sample app seems to quietly fail to discover any devices if its location permissions are denied by the OS. (Settings -> Privacy -> Permission Manager -> Location).  Took me a while to discover my problem! :)

douglasjunior, 2021-04-28:
Thanks @masonlee, this issue is already covered by README https://github.com/douglasjunior/AndroidBluetoothLibrary#known-issues.

masonlee, 2021-04-29:
Ah, I somehow missed that, @douglasjunior !  The issue seems to apply to classic as well; I've made a PR to add some clarity to that readme section, hopefully making it a bit more idiot-proof. :)