douglasjunior/AndroidBluetoothLibrary #25: status NONE after try to connect

Amirhgh74, 2018-06-06:
I am having a problem connecting to a paired device 
when I try to connect to the device , status changes to connecting and right after that it changes to NONE 

what should I do to fix this ? @douglasjunior 

im using classic mode and set the uuid to be null as you said for finding all the devices

a9336, 2018-09-29:
Hello,

I have the same problem. Maybe is the  <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />

the librarie uses to connect. but this permition is only for operating system app´s. so the library dont permit to connect. (android make this change on android 5.0)

you can connect if the device are rooted.

douglasjunior, 2018-09-30:
I've had similar problems on some devices, it works well on some smartphones, but not on others. I think it's some incompatibility with the hardware of the smartphone (depends on the version of Android and model of the device).

This library is only a bridge to the Android bluetooth API, and does not implement the Bluetooth protocol directly, so there is nothing that can be done in the library. (The Bluetooth Classic module is based on a [Google example](https://github.com/googlesamples/android-BluetoothChat/blob/master/Application/src/main/java/com/example/android/bluetoothchat/BluetoothChatService.java))

In my experiences, I had fewer problems with Bluetooth 4.0+ (Low Energy) than Bluetooth Classic. So I recommend using this version whenever possible.

douglasjunior, 2018-09-30:
@Amirhgh74 
> im using classic mode and set the uuid to be null as you said for finding all the devices
Is true when working with Bluetooth Low Energy.

With Bluetooth Classic the UUID is required.

closed 2018-12-15 by douglasjunior