douglasjunior/AndroidBluetoothLibrary #26: Envio duplicado

aviler, 2018-07-31:
Oi Douglas! Primeiramente queria te parabenizar por seu trabalho! 
Cara estou com um problema aqui em alguns momentos que aciono comando de envio de mensagem (writeln) acabo recebendo a mensagem de forma duplicada, até no logcat parece que sua lib faz o envio duplicado. Alguma chance de ser o fato de estar conectado a uma Characteristic com dois Descriptor, nesse caso o dispositivo BLE (HMSoft-10 lendo na serial) receberia uma mensagem em cada Descriptor?

jjrodcast, 2018-08-01:
@aviler pode colocar um Thread.sleep(1000) e visualizar novamente sua data. Acho que é o tempo de resposta obtida no Bluetooth.

douglasjunior, 2018-08-01:
Oi @aviler, não é problema na Characteristic, pois ele só conecta em uma. Já o LogCat, ele imprime um "log" em vários momentos durante o envio, mas não quer dizer que está enviando em duplicidade.

Você disse que o problema só ocorre em alguns momentos? Você já tentou identificar o que acontece de especial nestes casos?

aviler, 2018-08-01:
Já avaliei bastante antes de vir perguntar, aliás fico muito grato pelas respostas rápidas de vocês.

@jjrodcast colocar um Thread.sleep logo após a chamada de write? Vou tentar isso aqui.

@douglasjunior Então, oque me surpreende é essa variação. Tipo identifiquei que quando estou em modo debug ocorre mais vezes, ainda mais se coloco breakpoint perto da chamada. Outra coisa é que estou acionando o write dentro de um onFinish() de CountDownTimer. Vou procurar uma alternativa para o CountDownTimer e testar o Thread.sleep daí retorno aqui.

closed 2018-12-15 by douglasjunior