douglasjunior/AndroidBluetoothLibrary #34: Does it have any code to read back from the device ?...I can write using the bluetoothwriter but a reverse functionality does'nt seem to be available.

faizanmir, 2019-03-23:
Hi, I have used your library in one of my projects and seems to work fine except for one part , i dont seem to have any control with the data sent back from the BLE device ...

douglasjunior, 2019-03-23:
Hi @faizanmir , you need to control the received data with `characterDelimiter` and `bufferSize`.

faizanmir, 2019-03-23:
HI douglas ,Thanks for your prompt reply ,I am pretty new to android as well as ble ,Your library seems to simplify the otherwise tedious task of making the bluetooth connection ...Could your please point me to an implementation of the same ...that would be so much more helpful....

faizanmir, 2019-03-23:
I can see that there is a an over ridden method OnDataRead ....is this where the data that is read from the device can be accessed ...sorry for being such a noob

douglasjunior, 2019-03-23:
The library will retains the received data until find a `characterDelimiter` or fill the `bufferSize`. Then, the library will call the `onDataRead ` listener.

faizanmir, 2019-03-23:
could you please point me towards a resource for the characterDelimiter ,I cant seem to find much about it ,Your help is appreciated

faizanmir, 2019-03-23:
I am able to see the data back in logs 
but not in dataread

douglasjunior, 2019-03-23:
You need to find a pattern in the data received from your device. 

For example, if your data received is something like this:

```
1;some value;234;other valeu
2;some value;567;other valeu
3;some value;901;other valeu
```

And you specify the 'break line' `\n` as the `characterDelimiter`, then the `onDataRead` will receive:
- 1;some value;234;other valeu
- 2;some value;567;other valeu
- 3;some value;901;other valeu

If you specify the `;` as the `characterDelimiter`:

- 1
- some value;
- 234;
- other valeu
- 2
- some value
- 567
- other valeu
- ...

faizanmir, 2019-03-23:
V/BluetoothLeService: onCharacteristicChanged:              " av,9,,NOCON,~ "  is the data I receive however if I try to access the OnDataRecieved in the library nothing shows ....even in the logs ....

faizanmir, 2019-03-23:
TYSM i decreased the size of buffer and got the data ....tysm for your help

closed 2019-03-23 by faizanmir

douglasjunior, 2019-03-23:
The `~` indicated the end of the data? If yes, you can use `~` as the `characterDelimiter`.