douglasjunior/AndroidBluetoothLibrary #36: Read data from Bluetooth device

amarpulli544, 2019-05-14:
Hi @johnhowe @douglasjunior ,
I would like  try  to read the data from bluetooth device over android app.  When I was send  the data to bluetooth device over android app the data has been sent successfully. But at the same time when i was try to read the data at android app side, am unable to read the data. I am using HM-10 bluetooth module. Here mDevice is a bluetooth mac address.
This is my code
```
final BluetoothService service = BluetoothService.getDefaultInstance();
        //
        service.setOnEventCallback(new BluetoothService.OnBluetoothEventCallback() {
            @Override
            public void onDataRead(byte[] buffer, int length) {
                str = new String(buffer,0,length);
                runOnUiThread(new Runnable() {
                    @Override
                    public void run() {
                        Toast.makeText(getApplicationContext(),"received data="+str,Toast.LENGTH_SHORT).show();
                    }
                });
            }

            @Override
            public void onStatusChange(BluetoothStatus status) {
                Log.d(TAG,"status:"+status);
            }

            @Override
            public void onDeviceName(String deviceName) {
                Log.d(TAG,"device:"+deviceName);
            }

            @Override
            public void onToast(String message) {
                Log.d(TAG,"message:"+message);

            }

            @Override
            public void onDataWrite(byte[] buffer) {

            }
        });
        service.connect(mDevice);
```
Can anyone please help me. How to receive the data over the android app.
Thanks in advance
Amar Pulli.

douglasjunior, 2019-05-14:
You need to set a `bufferSize` and `characterDelimiter` to your config. And please, don't draw users attention to your comments.

amarpulli544, 2019-05-14:
@douglasjunior thanks for response,
This is my config
```
 BluetoothConfiguration config = new BluetoothConfiguration();
        config.context = getApplicationContext();
        config.bluetoothServiceClass = BluetoothLeService.class;
        config.bufferSize = 1024;
        config.characterDelimiter = '\n';
        config.deviceName = "BT05";
        config.callListenersInMainThread = true;
        config.uuidService = UUID.fromString("0000ffe0-0000-1000-8000-00805f9b34fb"); 
        config.uuidCharacteristic = UUID.fromString("0000ffe1-0000-1000-8000-00805f9b34fb"); 
        config.transport = BluetoothDevice.TRANSPORT_LE; 
        BluetoothService.init(config);
```

amarpulli544, 2019-05-15:
Hi @douglasjunior ,
I already set `bufferSize` and `characterDelimiter` to my config file. which is shown below. But still  am not able to receive the data.
```
BluetoothConfiguration config = new BluetoothConfiguration();
        config.context = getApplicationContext();
        config.bluetoothServiceClass = BluetoothLeService.class;
        config.bufferSize = 1024;
        config.characterDelimiter = '\n';
        config.deviceName = "BT05";
        config.callListenersInMainThread = true;
        config.uuidService = UUID.fromString("0000ffe0-0000-1000-8000-00805f9b34fb"); 
        config.uuidCharacteristic = UUID.fromString("0000ffe1-0000-1000-8000-00805f9b34fb"); 
        config.transport = BluetoothDevice.TRANSPORT_LE; 
        BluetoothService.init(config);
```
Android studio log file :

```
05-15 09:40:19.497 29878-29890/com.fueblabs.fueb.testbluetooth D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
05-15 09:40:19.499 29878-29890/com.fueblabs.fueb.testbluetooth D/BluetoothGatt: configureMTU() - device: 00:15:84:00:3C:01 mtu: 512
05-15 09:40:19.501 29878-29890/com.fueblabs.fueb.testbluetooth V/BluetoothLeService: requestMtu: true
05-15 09:40:19.502 29878-29890/com.fueblabs.fueb.testbluetooth V/BluetoothService: updateStatus() CONNECTING -> CONNECTED
05-15 09:40:19.504 29878-29878/com.fueblabs.fueb.testbluetooth D/BLE: device:BT05
05-15 09:40:19.504 29878-29878/com.fueblabs.fueb.testbluetooth D/BLE: status:CONNECTED
05-15 09:40:19.515 29878-29889/com.fueblabs.fueb.testbluetooth D/BluetoothGatt: onConfigureMTU() - Device=00:15:84:00:3C:01 mtu=23 status=0
05-15 09:40:19.516 29878-29889/com.fueblabs.fueb.testbluetooth V/BluetoothLeService: onMtuChanged: 23 status: 0
05-15 09:40:34.726 29878-29890/com.fueblabs.fueb.testbluetooth V/BluetoothLeService: onCharacteristicChanged: �
05-15 09:40:53.006 29878-29889/com.fueblabs.fueb.testbluetooth V/BluetoothLeService: onCharacteristicChanged: test
```
Here am able to see receive data in log file only. 
How can i fix this?
Thanks and regards
Amar

closed 2019-05-17 by amarpulli544