douglasjunior/AndroidBluetoothLibrary #40: How to read using the classic bluetooth?

michaelboeding, 2020-09-03:
How do I read using the classic bluetooth? Not seeing anything in the documentation. Thanks

douglasjunior, 2020-09-03:
You should use the `setOnEventCallback` and then `onDataRead`.

[Sample project](https://github.com/douglasjunior/AndroidBluetoothLibrary/blob/5b9061b7eb02fab3126e91621cedfaa0f7d8a28c/Sample/src/main/java/com/github/douglasjunior/bluetoothsample/MainActivity.java#L154).

michaelboeding, 2020-09-03:
Thanks for the quick reply! Im pretty new to Android/Java I have that implemented in my code but how do I get that to call. 

Like for example I use service.write(byteArray[]) to write to the bluetooth device which in turn calls the onDataWrite(). Is there a similar way to call the read? Or how would I do that?

Thanks again, really appreciate it.

douglasjunior, 2020-09-03:
You need to read more about Bluetooth and asynchronous programming.

Receiving data via bluetooth is asynchronous, so there is no method for reading synchronously.

michaelboeding, 2020-09-03:
Just did some reading and I see. I am running BLE(can't seem to use BLE on the android version because the bluetooth module has a bug) on my IOS version of the app I'm working on so I was thinking it was setup like that, where I would read whenever I needed a value. It's just weird that I'm not receiving any data from the callback, since the device should be sending data constantly when connected. Will do some more digging, thanks.

closed 2023-02-13 by douglasjunior