douglasjunior/AndroidBluetoothLibrary #41: onDataRead doesn't invoke

calypsom, 2021-02-28:
I'm struggling on reading data . I'm using RN4871 device and connecting and writing works like a charm, but can't get onDataRead invoke. I've been trying different characterDelimiters and bufferSizes and I understand how they should work, but I'm not able to receive anything.

My config code on main activity:
```
 BluetoothConfiguration config = new BluetoothConfiguration();
        config.context = getApplicationContext();
        config.bluetoothServiceClass = BluetoothLeService.class;
        config.bufferSize = 24;
        config.characterDelimiter = '\n';
        config.deviceName = "ble test";
        config.callListenersInMainThread = false;
        config.uuidService = UUID.fromString("49535343-FE7D-4AE5-8FA9-9FAFD205E455"); // Required
        config.uuidCharacteristic = UUID.fromString("49535343-1E4D-4BD9-BA61-23C647249616"); // Required
        config.uuid = null;
        BluetoothService.init(config);

        mService = BluetoothService.getDefaultInstance();
        mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter();

        deviceList.addAll(mBluetoothAdapter.getBondedDevices());

        mService.setOnEventCallback(this);
```
setOnEventCallback:
```
  @Override
    public void onDataRead(byte[] buffer, int length) {
        Log.d(TAG, "onDataRead");
    }

    @Override
    public void onStatusChange(BluetoothStatus status) {
        Log.d(TAG, "onStatusChange: " + status);
    }

    @Override
    public void onDeviceName(String deviceName) {
        Log.d(TAG, "onDeviceName: " + deviceName);
    }

    @Override
    public void onToast(String message) {
        Log.d(TAG, "onToast");
    }

    @Override
    public void onDataWrite(byte[] buffer) {
        Log.d(TAG, "onDataWrite");
    }
```

On AndroidManifest I have added location and bluetooth permissions except BLUETOOTH_PRIVILEGED, which seems to be protected permission.

What I'm missing?

douglasjunior, 2021-03-01:
`OnDataRead` will be called when a` characterDelimiter` is received, or when the total data received exceeds the character limit of `bufferSize`.

douglasjunior, 2021-03-01:
https://github.com/douglasjunior/AndroidBluetoothLibrary/issues/34#issuecomment-475864775

calypsom, 2021-03-02:
Ok! Any idea what it should be then?

douglasjunior, 2021-03-14:
The best option is to use some bluetooth scanner app to view the UUIDs available on your device.