douglasjunior/AndroidBluetoothLibrary #45: OnReadyRead only when bufferSize is reached

Shunt22, 2021-12-08:
As mentioned in #41 
`OnDataRead` will be called when a ` characterDelimiter` is received, or when the total data received exceeds the character limit of `bufferSize`.

_Originally posted by @douglasjunior in https://github.com/douglasjunior/AndroidBluetoothLibrary/issues/41#issuecomment-787958999_

Is there anyway to set OnDataRead to be called _ONLY_ when `config.bufferSize` is reached?
I don't care about character  delimiter.  Just want to ready fixed-size portions of data.

douglasjunior, 2021-12-08:
You can pass a `characterDelimiter` that never will be received by your device.

Shunt22, 2021-12-08:
> You can pass a `characterDelimiter` that never will be received by your device.

In my case I transfer pretty much random data bytes (audio data), so I can't for sure say which character will never appear. 
I tried '\n', '\0',0x00, etc as characterDelimiter, but these characters sometimes occur in data stream and OnDataRead is called.

closed 2023-02-13 by douglasjunior